The-OpenROAD-Project/alpha-release #3: "make export-tools" fails with "Unable to find image 'yosys:latest' locally"

RTimothyEdwards, 2019-07-17:
I got partway into the build, and seem to have successfully done "make build-tools" (which kept throwing errors that look like maybe a non-zero exit status indicating a successful exit, but that's for another issue).  However, "make export-tools" fails immediately with:

mkdir -p export
rm -rf export/yosys.tar.gz
id=$(docker create yosys) ; \
  docker cp $id:\build.tar.gz export/yosys.tar.gz ; \
  docker rm -v $id
Unable to find image 'yosys:latest' locally
Error response from daemon: pull access denied for yosys, repository does not exist or may require 'docker login'
must specify at least one container source
"docker rm" requires at least 1 argument.
See 'docker rm --help'.

Usage: docker rm [OPTIONS] CONTAINER [CONTAINER...]

Remove one or more containers

I am a novice with docker (just used it successfully for the first time a few days ago), and I do not know how to debug this.  I do know that 'docker login' should not be the problem because I was successfully logged in.

Any ideas?

tajayi, 2019-07-17:
Hi Tim,

Happy to have testers on our platform. We have integrated several tools from different teams and are still working on integrating them more tightly - hence the dockerized build process.
We currently advise the use of pre-compiled binaries that are available on the repo release and are validated on Redhat/CentOS 6/7.

mithro, 2019-07-17:
My guess is that `docker create yosys` failed in some way? It looks like `$id` didn't get a value...

RTimothyEdwards, 2019-07-17:
I am trying to build on a Debian system (specifically PureOS).  Not sure how well the RedHat precompiled build is going to work.  The prerequisites of Tcl/Tk 8.5, libreadline 6, and libffi 5 are quite out-of-date (I have Tcl/Tk 8.6, libreadline 7, and libffi 6).  I will give it a try.

tajayi, 2019-07-31:
We have added "runtime" docker images with the tools pre-installed. We will try to update this alongside the tool exports. in the long term, we are still looking to make improve the build.

See instructions at
https://github.com/The-OpenROAD-Project/alpha-release/tree/master/build#docker-install

closed 2026-09-09 by maliberty