The-OpenROAD-Project/alpha-release #6: Add Western Digital SWERV as test bench

mithro, 2019-07-17:
It would be awesome to add Western Digital SWERV RISC-V core as a test bench. You can find it at https://github.com/westerndigitalcorporation/swerv_eh1 and it is released under an Apache 2.0 license.

> Western Digital’s RISC-V SweRV core is a 32-bit in-order core featuring a 2-way superscalar design and a nine-stage pipeline. When implemented using a 28 nm process technology, the core runs at up to 1.8 GHz. As for simulated performance, the SweRV core delivers 4.9 CoreMark/MHz, which is a bit higher when compared to ARM’s Cortex-A15. Western Digital will use its RISC-V cores for its own embedded designs, such as flash controllers and SSDs.

[image]

[image]

Only issue is SWERV is *just* a CPU core and not a whole SoC?

tajayi, 2019-07-17:
Thanks Tim. We'll try to queue this up

tajayi, 2019-08-09:
swerv is also now running through the flow. I still need to do some characterization on it

ZvonimirBandic, 2019-08-09:
wow, this is really awesome! Tim told me today in email, I was not aware. I am interested to learn more about this, and potentially reproduce what you are doing. Let me know if you found out any issues.

tajayi, 2019-09-09:
Swerv is now mostly functional through the flow on 2 platforms. There are some DRCs issues that will hopefully be resolved by on-going efforts around DRCs and QoR.

**swerv_nangate45**
517 Viols
Placement Density:9.75%(196620/2016361)

**swerv_tsmc65lp**
30 Viols.
Placement Density:37.97%(762620/2008285)

**swerv_wrapper_tsmc65lp**
43 Viols.
Placement Density:41.69%(789823/1894629)
Several connectivity issues: Narrow Channels (Placement), Missing std cell rails

closed 2019-09-09 by tajayi