The-OpenROAD-Project/alpha-release #7: Add the LowRISC Ibex SoC as test bench

mithro, 2019-07-17:
It would be good to add the LowRISC Ibex core + SoC as a test bench. You can find it at https://github.com/lowRISC/ibex

> This core was initially developed as part of the PULP platform under the name "Zero-riscy" [1], and has been contributed to lowRISC who maintains it and develops it further. It is under active development, with further code cleanups, feature additions, and test and verification planned for the future.

[image]

The core has been taped out multiple times and will have a strong focus on adding and improving security -- hence they care deeply about transparency and reproducibility. It would be an awesome fit for an open source toolchain!

tajayi, 2019-08-07:
Ibex has been added and is now running through the flow as of the latest commint. There are about 8 DRCs on nangate45 and about 7 DRCs on tsmc65lp. We'll continue working (on a broader scope) to get lvs clean consistently across all designs

mithro, 2019-08-08:
Great work!

How do I replicate your results?

tajayi, 2019-08-08:
You should be able to run it on the Nangate45 platform.
Follow the instructions on the flow page: https://github.com/The-OpenROAD-Project/alpha-release/tree/master/flow (Let me know if it needs more work)

Once the flow is setup (you may want to run the default small design to validate your isntalld), open the Makefile in the `flow` folder and uncomment ibex_tsmc65lp to set it as the design. Run `make`

closed 2019-08-14 by tajayi