The-OpenROAD-Project/alpha-release #8: Explore SystemVerilog support using Verilator frontend

tajayi, 2019-07-17:
Converting System Verilog to plain Verilog using .xml dump from Verilator.
See project at: https://github.com/jrrk/sysver2ver

tajayi, 2019-09-27:
Also see: https://github.com/zachjs/sv2v

NilsGraf, 2019-09-27:
@zachjs @hzeller @mithro

FYI, the latest version of [sv2v](https://github.com/zachjs/sv2v) is able to convert the Ibex core from SystemVerilog to Verilog and Yosys can read it in, see [this PR](https://github.com/lowRISC/ibex/pull/342).

Would it make sense to include sv2v in the OpenROAD flow?

tajayi, 2019-12-02:
For now we are choosing to have the flow start at "Verilog" -> "GDS". We are currently recommending the following tools to help

https://github.com/bespoke-silicon-group/bsg_sv2v
https://github.com/zachjs/sv2v

closed 2019-12-02 by tajayi

mithro, 2019-12-04:
You should also check out https://GitHub.com/SymbiFlow/sv-tests and https://SymbiFlow.github.io/sv-tests to see what level of SystemVerilog open source tools support.