The-OpenROAD-Project/alpha-release #11: Many verilog files for a design

Colin-Holehouse, 2019-07-19:
In the case where a design is made up of many verilog files, it would be beneficial to be able to use the .vc package files that are used to load the design into simulation tools - this would simplify setup and ensure a consistent set of files are used.

This is the approach that we use in ARM to load verilog files for synthesis

tajayi, 2019-08-08:
The .vc file is a Verilog Command file intended for Synopsys Simulator.

 I don't think any other tool supports reading vc files. It is probably not advisable to support it in OpenROAD either since we may not fully support all the flags/parameters the file format supports.

It is fairly straight forward to adapt it into the current design.mk file.
- verilog files will explicitly need to be passed to the flow using `VERILOG_FILES`
- No support for `-y` which searches the specified paths for sources to verilog modules in the design. Full file paths will be required
- `+incdir` can be specified in the `VERILOG_INCLUDE_DIRS` variable. (Supports multiple paths)

closed 2019-08-08 by tajayi

Colin-Holehouse, 2019-08-09:
The .vc file is used by all simulators as far as I know.

We also use it to read the RTL into genus with the command
read_hdl -f <vc_file>

Support of the vc_file format in synthesis tools is somewhat patchy so this approach may not work for all IP [even if it does work for ARM IP]