The-OpenROAD-Project/alpha-release #12: Need to be able to apply verilog parameter values to top level module

Colin-Holehouse, 2019-07-20:
If the top level module has paramters that need to be modified from their default, then there needs to be a mechanism for the user to specify the parameters to be applied. 

Suggestion:
- Define the parameters in the designs/<design>.mk file e.g.
export RTL_PARAMS = \
  L2_CACHE 1 \
  NEON_FP 0 \
  CRYPTO 0 \
  CPU_CACHE_PROTECTION 0

- Add the following into scripts/synth.tcl
if {[array names ::env RTL_PARAMS] != ""} {
  select -module $::env(DESIGN_NAME)
  dict for {key value} $::env(RTL_PARAMS) {
    setparam -set $key $value
  }
}

tajayi, 2019-08-09:
This has been added to the flow in commit 9a97a21ae5f3a81899b3ede3628c693a8d64e2b3

Add the following to the design config
```export VERILOG_TOP_PARAMS = D_WIDTH 64 A_WIDTH 16```

closed 2019-08-09 by tajayi

Colin-Holehouse, 2019-08-09:
Thanks - this looks to be working correctly