The-OpenROAD-Project/alpha-release #13: Coping with verilog include directory settings

Colin-Holehouse, 2019-07-20:
Need a mechanism to be allow verilog includes to be picked up from a set of include directories

Tried to set this up by adjusting scripts/synth to be:
foreach file $::env(VERILOG_INCLUDE) {
  read_verilog -Idir $file
}

but subsequent read_verilog calls dont cant pick up the required include files - any idea what the right thing to do here is? Use verilog_defaults? Use -Idir options for every read_verlog command?

Testcase: ca53_cpu_config1

tajayi, 2019-08-08:
This has been added to the flow in commit e1d3bc6195a067e271f0f6c26327ffcc4adf02f4. See an example of using it at https://github.com/The-OpenROAD-Project/alpha-release/blob/master/flow/designs/jpeg_tsmc65lp.mk#L5

I think the trick is to add multiple "-Idir" for each file read

closed 2019-08-08 by tajayi

Colin-Holehouse, 2019-08-09:
Thanks - this is working at my end as well.