The-OpenROAD-Project/alpha-release #15: Several questions about running testcases

Devilbilly, 2019-07-25:
Hi, 
I tried to ran all testcases in Makefile, but only a few of them can be run successfully.
I have recorded the problems I encountered below, please tell me if I am doing it wrong or I missed something.

Thanks.

1. 
$> make DESIGN_CONFIG=./designs/gcd_freepdk45.mk 

ERROR: TCL interpreter returned an error: can't read "::env(TIEHI_CELL_AND_PORT)": no such variable
make: *** [results/gcd/synth.yosys.v] Error 1

2. 
$> make DESIGN_CONFIG=./designs/gcd_freepdk45.mk
$> make DESIGN_CONFIG=./designs/gcd_nangate45.mk

/bin/bash: line 17: 30955 Segmentation fault      (core dumped) RePlAce -bmflag etc -lef ./tech/gcd/merged_padded.lef -def ./results/gcd/floorplan.def -verilog ./results/gcd/floorplan.v -lib ./platforms/nangate45/NangateOpenCellLibrary_typical.lib -sdc ./results/gcd/floorplan.sdc -output ./results/gcd/replace -t 1 -timing -resPerMicron 1.59 -capPerMicron 0.235146e-12 -skipIP -plot -experi output -den 0.509 -initCoef 0.00002 -onlyGP 2>&1
     30956 Done                    | tee ./logs/gcd/RePlAce.log
make: *** [results/gcd/place_gp.def] Error 139

3. ./designs/src/dynamic_node/design_fixed.v should be ./designs/src/dynamic_node/design.v in dynamic_node_nangate45.mk?

4. ./platforms/tsmc65lp/ not found (will you provide this?)

5. Router is unable to finish in ibex_nangate45.mk
[CGRouter::Routing]:rerouting started OF:27021 WL:378848 VIA:260735=1161053 (1037.7 MB) at Thu Jul 25 23:27:28 2019
[CGRouter::Routing]:*  0/999 P= 3 make: *** [results/ibex_core/route.guide] Terminated

tajayi, 2019-07-26:
Can you please try the latest version?

1. The freepdk45 platform (and all tests) were explored but abandoned in favor of nangate45. These are not expected to go through the flow. I have deleted the freepdk45 platform and tests

2. Please run 'make clean_all' when switching between tests running on the same platform. This should resolve this

3. Fixed

4. We do not plan to provide any tsmc65lp kits/informaiton. These are private

5. Please pull the latest release and binary installer. We have migrated to a new global router.
I just validated that aes_nangate45 and gcd_nangate45 work. We prioritize support for tsmc65lp and are still resolving some routing bugs for nangate45. This currently affects the ibex_nangate45 and dynamic_core_nangate45 designs. I'll leave the issue open as we continue to look into it

Devilbilly, 2019-07-26:
Thanks for fast response.
I have pulled lastest version and it seems that FRlefdef is missing.
                ____     ____   ___  _   _ _____ ___ _   _  ____
               | ___|   |  _ \ / _ \| | | |_   _|_ _| \ | |/ ___|
               |___ \   | |_) | | | | | | | | |  | ||  \| | |  _
                ___) |  |  _ <| |_| | |_| | | |  | || |\  | |_| |
               |____(_) |_| \_\\___/ \___/  |_| |___|_| \_|\____|

^[[0menvsubst < ./scripts/fastroute.rsyn > ./objects/dynamic_node_top_wrap/fastroute.rsyn
FRlefdef --no-gui --script ./objects/dynamic_node_top_wrap/fastroute.rsyn 2>&1 | tee ./logs/dynamic_node_top_wrap/5_1_FRlefdef.log
/bin/bash: FRlefdef: command not found

tajayi, 2019-07-26:
You may need to open a new shell and source the setup.sh file again. I just verified that it should be extracted to `openroad\OpenROAD-2019-07-25_12-54\bin\`

Devilbilly, 2019-07-26:
Open new shell and source again solve the problem.

I can now run aes and gcd successfully, while ibex and dynamic_node are still unable to be routed.

Thanks.

tajayi, 2019-08-09:
ibex and dynamic_node should now be running fine. Thew new swerv_nangate45 is still being debugged.

closed 2019-08-09 by tajayi