The-OpenROAD-Project/alpha-release #17: Restate OpenROAD supported Linux distributions to be more forward looking

oharboe, 2019-08-05:
Currently OpenROAD states that only CentOS 6 is supported. Issues raised with other distributions and operating systems are consequently closed.

My concern with this is that CentOS 6 nine years old and it stopped getting full updates in 2017 and is scheduled for no more maintenance updates in November 2020: https://en.wikipedia.org/wiki/CentOS#End-of-support_schedule

**I propose** that OpenROAD restates it's supported Linux distributions as something along the lines of:

 _"Currently OpenROAD is being tested with a CentOS 6 Docker environment baseline, but we would like to move to a newer Linux distribution as a baseline. We also welcome patches to improve support for other Linux distributions and operating systems as long as they do not overly complicate the build process."_

**If OpenROAD** continues to have the position of only supporting CentOS 6, then that's going to alienate a lot of potential testers and users and it is going to make it highly impractical to test out OpenROAD outside the limited context of the provided docker images.

**If the OpenROAD** supported Linux distributions position is restated along the lines above, then we should see more people interested in trying out OpenROAD and consequently a rise in number of pull requests.

tajayi, 2019-08-05:
Thanks @oharboe  for all the testing you've been doing. I have updated the readmes.

 We are not insisting on CentOS6, but it just happened to the be first OS we have been testing and validating. CentOS 7 has also been validated.

In future, we plan to support newer and arbitrary Linux-based systems. There are some short term plans to address pressing issues and in the long term we are looking to consolidate many of the tools and make them more cohesive.

closed 2019-08-05 by tajayi

oharboe, 2019-08-05:
super!

I believe this simply  needs to be cpmmunicated more widely and prominently.

tajayi, 2019-08-05:
Thanks. We were initially trying to deter people from going down the uncharted rabbit hole of trying to compile and build every binary in random environments (that we haven't tested).
Our goal is definitely to enable that to at least a reasonable degree