The-OpenROAD-Project/alpha-release #18: Only nangate45 is included in the alpha-release project

oharboe, 2019-08-05:
If you follow the instructions at https://github.com/The-OpenROAD-Project/alpha-release/tree/master/flow it says "There is an initial platform setup required in order to use the OpenROAD flow. This has already been performed for the FreePDK45, Nangate45 and TSMC65(restricted) platforms. See sample platforms setup in the platfomrs directory.", but when you clone alpha-release you'll find that the alpha-release master only contains "flow/platforms/nangate45"

I'm interested in the the tsmc64lp process w/ARM memories as I'm inferring small memories in my design(yosis identifies the memory inference corectly), but these memories are instantiated as flip-flops, therefore synthesis pretty much takes forever(I stopped it after 24 hours).

tajayi, 2019-08-05:
We have removed the "FreePDK45" platform (updated readme) in favor of Nangate45 which is more complete.

Unfortunately the ARM TSMC65LP platform (tech files, lib, memory compilers, etc) are restricted. If you have any existing enablements, you can attempt to add a new platform, however our tools are untested on it

We have a very preliminary "fake" memory generator that we are working on for nangate45. It will only generate lefs and libs for use with the flow. There will be no actual gds. We did some preliminary exploration with OpenRAM and the lefs generated were not very amenable for routing (even innovus couldn't route cleanly). If you're familiar with LEFs, my advise would be to fake one up.

oharboe, 2019-08-05:
will this mock(a term of art for a placeholder for tye real thing) memory
generator work in conjunction with yosis memory inference just as  fpga
tools do?

i am afraid that i have no experience with lef files, i am out of my depth
there.

> We have removed the "FreePDK45" platform (updated readme) in favor of
> Nangate45 which is more complete.
>
> Unfortunately the ARM TSMC65LP platform (tech files, lib, memory
> compilers, etc) are restricted.
"restricted" is a quite overloaded word. do you mean that it is under NDA?

you have any existing enablements, you can attempt to add a new platform,

tajayi, 2019-08-05:
In typical ASIC flows, a developer is responsible for:
1. identifying all large memories in the design
2. running a memory compiler to generate the memory views (lef, lib, gds, cdl, etc)
3. Updating the code/design to use the memory modules generated. (Will be passed to the synthesis tools as a .lib or .v file)

Yosys (or any other ASIC synth tool) will not automatically map/generate memory macros based on behavioural RTL.

The nangate45 is not a real/fabricatable platform anyway so going all out to generate a "correct" gds is not necessary. A lib file will provide timing information and a lef file will provide physical information for the tools. I'll check the status of the generator with another contributor who was working on it.

> "restricted" is a quite overloaded word. do you mean that it is under NDA?

By restricted I mean it is only available to people who have signed an agreement with Arm and have obtained the necessarily files directly from Arm. There are also NDAs associated with the agreement. A lot of our work was solely targeted at this node, however, we added nangat45 support to at least share something with users that don't have that agreement

oharboe, 2019-08-05:
Thanks for the clearification!

oharboe, 2019-08-05:
is there an example of such a LEF to  mock memories?

tajayi, 2019-08-05:
You can find some example here from OpenRAM
https://github.com/VLSIDA/OpenRAMLibrary

mguthaus, 2019-08-05:
We have fixes that are soon to be released for OpenRAM LEF files There were
a few bugs relating to blockages, but those were easily fixed.

Matt

mguthaus, 2019-08-05:
By the way, if there are bugs, report them! This is what open source is...
You should give back to the community.

Matt

oharboe, 2019-08-05:
Thanks! Looking forward to trying it out.

I've reported half-dozen issues reproted and had two trivial pull requests
merged so far. I'm just finding my way around the project currently.

mguthaus, 2019-08-05:
I meant that they found an issue with OpenRAM but didn't report it.... :)

oharboe, 2019-08-05:
Ah, yes. Good form is to not just work around bugs in open source projects
one depend on, but to report bugs and contribute back bugfixes. It's real
work, but a great way to pay back.

tajayi, 2019-08-06:
@mguthaus Sorry about that. I had a short list of things I was keeping track of but it got lost in the shuffle to trying to get the release out the door (especially when we realized it may not be viable for the timeframe).

A few things I remember when I was attempting to use the generated sram in design compiler and innovus:
1. Inconsistent naming of pins across LEF / lib / verilog (I think it had to do with capitalization of ports). Caused the tools to barf.
1. In the lib file voltage_unit needs to be "1V" instead of "1v"
1. For some reason (I don't quite remember) I wasn't able to use the lef as is. I had to re-export a lef abstract from virtuoso.
1. Hard to access pins in the lef (not on the edges and I think there was a blockage). Innovuse couldn't route it.
1. [Feature Request] Support for multi-port rams

Unfortunately, it'll take some time for me to re-spin the environment so I can't add more color to these issues. Let me know if you'd like me to file an issue on the page.

Thanks

mguthaus, 2019-08-06:
>    2. In the lib file voltage_unit needs to be "1V" instead of "1v"
>    3. For some reason (I don't quite remember) I wasn't able to use the
>    4. Hard to access pins in the lef (not on the edges and I think there
>    5. [Feature Request] Support for multi-port rams

Thanks for the summary, Tutu. If you do find more, please file a bug on our
GitHub page:
https://github.com/VLSIDA/OpenRAM
I believe most of these are resolved in versions over the last year.

1. Fixed. We ran into this as well.

2. Fixed. We've also tested .lib with multiple STA tools now, some had this
restriction in the parser.

3. Not sure about this, but we are using it fine in several tools. If you
have more information, that would be great.

4. This was a blockage export problem in the leaf cells that is fixed.

5. We released multi-port (1 or 2 port only) earlier this year.

tajayi, 2019-08-06:
WIll do. I'll give it another spin

closed 2019-08-06 by tajayi

oharboe, 2019-08-07:
> In typical ASIC flows, a developer is responsible for:
> 
> 1. identifying all large memories in the design
> 2. running a memory compiler to generate the memory views (lef, lib, gds, cdl, etc)

@tajayi  Is there a fundamental reason why invoking the memory compiler can't be done automatically based on how Yosis inferred the memories?

Being able to automatically invoke the memory compiler based on inferred memories seems like a good way to increase automation and reduce the complexity of the workflow.

I'm sure manual override is required in some cases, but especially for design space exploration, it's nice to be able to have a workflow that out of the box connects the dots from Verilog to GDS and reports.

tajayi, 2019-08-07:
> @tajayi Is there a fundamental reason why invoking the memory compiler can't be done automatically based on how Yosis inferred the memories?

**Short Answer:** No
**Long Answer:** Mapping memories in an ASIC flow can get quite challenging. Several factors would significantly affect the usability and floorplan of the design. Items to consider would be:
1) Bank strategy (i.e one large ram vs several to access in  parallel)
2) Choosing between SRAM vs RF or other flavors of the SRAM generator available
3) Determining the number of readers and writers. Also whether they are in the same clock domain
4) Number of muxes in the SRAM 
5) SRAM compiler limitations (e.g. number of bits, number of words, write masks)
From my experience, most RTL designs have to be adapted to the type of SRAM available (instead of the other way around). Robust designs put minimal requirements on SRAM memory interface. 

FPGA tools like Vivado can automatically infer Block RAMs from RTL (as long as you follow a given template). In this case, the tools are well aware of the block ram structures in the device. For asic flows, it is quite challenging for the synthesis tools (i.e yosys) to adapt to arbitrary SRAM generators on arbitrary technology nodes.

With that being said, the end goal of OpenROAD is to have a "no human in the loop" flow where memory could potentially be identified and autogenerated. We are still a distance away

oharboe, 2019-08-07:
> With that being said, the end goal of OpenROAD is to have a "no human in the loop" flow where memory could potentially be identified and autogenerated. We are still a distance away

It's a pretty ambitious goal. A big step forward would be to generate RAMs where Yosis is able to infer and e.g. OpenRAM is able to provide.

Xilinx and Altera has shown the value of inferring from given patterns because logic emulation and implementation are guaranteed to match, there's a single source, so no chance of a difference between simulation and instantiated RAM.