The-OpenROAD-Project/alpha-release #22: Improve debugabilitiy of alpha-release

oharboe, 2019-08-06:
I'm currently experiencing a crash with alpha-release in FRlefdef, but I don't know how to debug it:

1. The FRlefdef included the alpha-release Docker image contains no debug information, nor is the source included.
2. Trivial modules work, it's impractical without a deep understanding of the system how to reduce this to a small example
3. The output from FRlefdef is uninformative, see below.
4. What I'm testing OpenROAD on is proprietary, so at best I could provide an obfuscated somewhat scaled down version.

```
# FRlefdef --no-gui --script ./objects/nangate45/LEArrayScanChain/fastroute.rsyn
[deleted]
LV routing round 2, enlarge 20 
1 threshold, 20 expand
total Usage   : 2368335
Max H Overflow: 23
Max V Overflow: 2887
Max Overflow  : 2887
Num Overflow e: 6277
H   Overflow  : 3494
V   Overflow  : 231494
Final Overflow: 234988

LV Time: 3.870000 sec
updateType 1
iteration 1, enlarge 9, costheight 34, threshold 0 via cost 0 
log_coef 0.223050, healingTrigger 0 cost_step 30 L 0 cost_type 1 updatetype 1
initial grid wrong y1 x1 [0 0] , net start [10 0] routelen 305
 checking failed 0
Segmentation fault (core dumped)

```

oharboe, 2019-08-08:
After investigation, the above problem is due to the design having thousands of top-level pins. The particular bug in FRlefdef is not relevant to this bug-report.

The problem is that upon failure, the logs should have some helpful output to create a bug-report.

Ideally:

- a clear error message
- stack-trace

tajayi, 2019-08-08:
Out of cuiousity, did ioPlacer generate any warnings or errors?

oharboe, 2019-08-08:
That information is now lost.

Are there any guidelines on how to report a bug and what information to
include?

tajayi, 2019-08-09:
I'd say as much as you an share is welcome. Errror messages, logs, desings, etc.
Was mostly curious for filing an issue to the ioPlacer team to tell them what it does vs what it should be doing.

Ps. Also renaming the ticket

closed 2026-09-09 by maliberty