The-OpenROAD-Project/alpha-release #24: Community communication space

OrionBytes, 2019-08-07:
1. OpenROAD has quite a big [team](https://theopenroadproject.org/our-team/) but not all participants are active GitHub users as I understand.

May it be useful to create Gitter community (like Chisel has) [https://gitter.im/freechipsproject/chisel3](https://gitter.im/freechipsproject/chisel3)
or Mattermost (if it's better to host locally, I can help with that) room to ask instant questions and help each other?

It may help newcomers to get in the project faster, most popular questions can be separated in Q&A, even before detailed docs. Github issues and pull requests can be integrated with notifications in a separate room.
Usual mail group can work in the same way, but technical chat room is usually faster from my personal experience.

2. There is a [list of current issues and limitations](https://github.com/The-OpenROAD-Project/alpha-release/blob/master/flow/docs/Known%20Issues%20and%20Limitations.pdf)

Maybe it'll be useful to create GitHub issues for those things? Hopefully, community will be able to check and try to fix it in particular cases.

3. Web interface

Before full IDEA cloud will be ready, maybe alpha version with examples can be integrated in existing [EDA playground](https://www.edaplayground.com/)? 

4. Regarding the pull requests

If it can save time for OpenROAD maintainers - general recommendations of how to form, check and describe contributions can be a good addition to README

---
If any idea is interesting - I'm open to help with that.

tajayi, 2019-08-08:
Hi @OrionBytes,
Thanks for your support and interest in OpenROAD.
1. We made an attempt to setup a community [here](https://community.theopenroadproject.org/login) but I don't think that picked up as well as we hoped. I'll discuss this with @tspyrou, @abk-openroad and @abdelrahmanhosny 

2. We are still working on a way to figure out project management and listing tasks/issues while making them transparent to potential contributors. CC'in @tspyrou 

3. EDA playground currently only performs synthesis and they have already integrated the same synthesis tool we use (i.e. yosys). We have something similar in the works. You can give it a spin [here](https://flow.theopenroadproject.org). @abdelrahmanhosny can provide more details. 

4. This is one of the top priority issues we are currently working on a scheme to provide unit tests, better builds and documentation for developers.

tajayi, 2019-08-09:
1.) Gitter created: https://gitter.im/The-OpenROAD-Project/community

2 & 4.) Still working on project management database The continous integration framework is still in progress but I've added a "Contributing" section to the readme. We will update that once that's ready. Staty tuned!

closed 2019-08-09 by tajayi