The-OpenROAD-Project/alpha-release #26: Add license information and other details for flow/designs/src/*

asb, 2019-08-08:
It would be great to add more info to flow/designs/src/*. At a minimum - ideally a link to the upstream project, the commit hash used to produce the .v, and ideally more info about how it was generated (e.g. what options were used). You should also import a copy of any license text and other attribution information, as most open source licenses require this.

[Really pleased to see you using Ibex as a benchmark for OpenROAD by the way - and looking forward to finding time to give the flow a go].

tajayi, 2019-08-08:
Thanks for the feedback. I'll try to add this. Some designs needed to be "elaborated" using a different tool because Yosys doesn't currenty support system verilog so we may not be able to have it in it's original form.

But we should be able to provide trace-ability, license and information to how the verilog sources files were generated.

tajayi, 2019-08-09:
I've started going through some of the designs and adding the requested information. I've done so for ibex. Let me know if this looks good.

asb, 2019-08-09:
Information on commit hash, how it was produced etc is a huge improvement - thanks! I'm not a lawyer, but including the original license also seems like a big step forward there. Thanks for such a rapid turnaround.

closed 2026-09-09 by maliberty