The-OpenROAD-Project/alpha-release #27: Add a place in alpha-release where it's easy to create pull requests for test-cases

oharboe, 2019-08-08:
Perhaps we can update alpha-release/flow/Makefile so that it has a "tests" target?

To meet it's ambitious goal, OpenROAD needs to grow a library of thousands of integration test-cases that are as small as possible to illustrate a particular problem.

Each test-case should have a few files as input and a way to check expected result.

It's very important that it is very easy to create these test-cases as there's going to be a very large number of them and they are going to be created as pull requests by regular users as they run into problems.

tajayi, 2019-08-09:
Again agreed with your point. The "alpha-release" is currently a short term solution to a proper release repo but I'm open to accepting more testcases. If you send me a pull request for a few designs, I'll add it to the repo. It currently scales to maybe tens of tests but not thousands. 

Currently, my "amateur" test/integration method is to run  ` make DESIGN_CONFIG=./designs/{design}.mk`.  I use gnu parallel to distribute it across a bunch of machines (see flow/parallelScript.txt). I'll start pushing the logs for the current tests so they can be diffed

We hope to have a more robust continuous integration framework soon.

oharboe, 2019-08-09:
I've kicked it off with https://github.com/The-OpenROAD-Project/alpha-release/pull/28

tajayi, 2019-08-09:
Thanks.. Looking at the pull request now.

Closing this ticket an looking at that

closed 2019-08-09 by tajayi