The-OpenROAD-Project/alpha-release #35: Include build log in release artifacts

oharboe, 2019-08-17:
It would be great to see how the output of the build in the alpha-release.

From this we could study e.g. which tests pass/fail, changes in quality of results, etc.

tajayi, 2019-09-05:
Can you please elaborate?

In the release tar.gz, there are breadcrumb files that inform you the version of the tools included. E.g

```
$ cat $OPENROAD/cts.version
Build Timestamp: 2019-08-15_13-07-44
7232ada8b7e88f0e14671cc6f9633c0dbe6d63e1
$
```

oharboe, 2019-09-05:
I would like to see the summary of the output of the tests that were run as part of the release. Which tests passed, which failed. A html link we can click on alongside the downloadables would be neat.

closed 2026-09-09 by maliberty