The-OpenROAD-Project/alpha-release #41: Problems launching magic gui

oharboe, 2019-08-30:
Normally, I can launch a GUI from within Docker on my desktop by running the command below, but I get an error message.

I run Ubuntu 18.04.

```
$ docker run --volume="/run/user/$UID/gdm/Xauthority:/root/.Xauthority:rw" --env="DISPLAY" --net=host --interactive -t openroad/alpha-release
[root@davos /]# magic
Error in startup script: 
    (default value for "-font" in widget ".text")
    invoked from within
"text $con -wrap char -yscrollcommand [list $w.sy set]  -foreground $COLOR(stdin)  -insertbackground $COLOR(cursor)"
    (procedure "InitUI" line 16)
    invoked from within
"InitUI $title"
    (procedure "::tkcon::Init" line 243)
    invoked from within
"::tkcon::Init"
    invoked from within
"if {![info exists ::tkcon::PRIV(root)] || \
	![winfo exists $::tkcon::PRIV(root)]} {
    ::tkcon::Init
}"
    (file "/openroad/OpenROAD-2019-08-15_12-11/lib/magic/tcl/tkcon.tcl" line 5327)
```

tajayi, 2019-09-05:
@RTimothyEdwards Any quick pointers you can think of? Not sure if it's related to the changes I made.

I'll continue looking into this

RTimothyEdwards, 2019-09-05:
I don't even see a specific error listed "(default value for "-font" in widget ".text")" does not sound like an error.  My best guess is that the container is missing fonts for Tk, meaning a problem with the Tcl/Tk install.  But Tk should be able to fall back on X11 fonts, and it would be surprising to find that the container didn't have X11 fonts installed somewhere. . .

RTimothyEdwards, 2019-09-05:
FYI, the error occurs inside the "tkcon.tcl" script, which is magic's console window in interactive mode under the Tcl/Tk interpreter.  Since you have been using magic in batch for OpenROAD applications, you may not have noticed if anything was wrong with the interactive version.  Technically, I would not suppose that OpenROAD is under any obligation to support interactive use of Magic within OpenROAD, but it would be nice to have that ability.

tajayi, 2019-09-05:
Thanks for the hints. I'll do some more debugging.
I have been able to run the GUI as packaged on CentOS 6/7. I wonder if this something to do with docker or the docker image configuration.

@oharboe Does the docker image have tk installed? Also is the docker image running ubuntu 18 or is it the docker host.

oharboe, 2019-09-05:
the docker image is openroad/alpha-release, so you probably know what it has installed.

i am running the command from nondocker ubuntu 18.04 desktop.

ohAitch, 2020-11-30:
Ran into this, got past it by [via](http://dba-amsterdam.blogspot.com/2015/03/hammerdb-setup-issues.html) `yum install xorg-x11-fonts*`, now have magic successfully running in Docker [on macOS](https://gist.github.com/cschiewek/246a244ba23da8b9f0e7b11a68bf3285#gistcomment-3477013)!

[image]

(Trying to compile natively met [problems](https://github.com/NixOS/nixpkgs/issues/105310), also related to X. )

closed 2026-09-09 by maliberty