The-OpenROAD-Project/alpha-release #43: Consider to use Alpine Linux for containers

XVilka, 2019-10-31:
[Alpine Linux](https://alpinelinux.org/) is a security-oriented, lightweight Linux distribution based on musl libc and busybox. It became popular for creating containers due to the very small size of resulting containers. Moreover, if you use Docker's [multistage build](https://docs.docker.com/develop/develop-images/multistage-build/) you can just copy the resulting binaries to the image, thus limiting the the end size to less than 50Mb. See more on the official [Alpine Linux Docker](https://github.com/alpinelinux/docker-alpine) repository.
Just a small comparison:
```
REPOSITORY  TAG     IMAGE ID      CREATED         SIZE
alpine      latest  cbc475bbc33b  55 seconds ago  5.53MB
ubuntu      latest  d131e0fa2585  7 days ago      102MB
debian      latest  2d337f242f07  5 weeks ago     101MB
centos      latest  9f38484d220f  7 weeks ago     202MB
```
See step-by-step article on [how to speed up your containers using Alpine Docker multistage builds](https://dev.to/fgriberi/speedup-your-docker-images-with-multi-stage-builds-1ipd).

tajayi, 2019-12-02:
CC'in @rovinski @abdelrahmanhosny

abdelrahmanhosny, 2019-12-02:
Hi @XVilka 

Thanks for the suggestions. 

The use of Alpine Linux is not relevant for OpenROAD use-case for the following reasons:
* We target CentOS as the main platform used for EDA tools for a long time. We don't want to test our build/test environments on other distributions.
* Image size is not a concern as we are not deploying 10s or 100s of containers in a web application. Containers are mainly used to ensure the integrity of the build and test environment (i.e. replication purposes).

Multi-stage builds have been adopted in actively maintained repos. See [OpenDB](https://github.com/The-OpenROAD-Project/OpenDB) and [OpenROAD](https://github.com/The-OpenROAD-Project/OpenROAD)

closed 2019-12-02 by abdelrahmanhosny

rovinski, 2019-12-03:
Maybe to clarify our goals, I don't think we want to imply that we never plan on supporting other containers / operating systems. I think compatibility is a good goal for the project. That being said, it is not currently on our roadmap because we have many other sponsor-directed milestones ahead of us. We may add compatibility goals once more of the necessary features have been developed.