The-OpenROAD-Project/alpha-release #47: Flow runs very well - Good to have 2 more features

kunalg123, 2019-11-12:
Hi
I ran entire swerv design using nangate 45nm, which has a post-synth instance count of about 140k...Runs extremely well.....Kudos to entire team

2 things I found missing was

1) PG mesh - Is there a way to built a rough PG mesh at "make floorplan" stage itself?
2) Fillers - Is there a way that at "make finish" stage, it could fill the design with FILLCELL? 

Thanks

tajayi, 2019-11-12:
Thanks @kunalg123 
At this time, the PG Mesh is only supported on the tsmc65lp platform. Future release will have a more robust PG mesh generation tool that will support more platforms. Fill cell insertion is also in the works for future versions. Please stay tuned

kunalg123, 2019-11-12:
Thanks
Should I keep this open to track the updates?

tajayi, 2019-11-12:
It would be preferable to close the issue, since it's being tracked seperately. Hopefully in the next few days we can publish the goals for the next few releases (jan and july).

closed 2019-11-13 by kunalg123