The-OpenROAD-Project/alpha-release #48: Trying new design on nangate45, routing fails

kunalg123, 2019-11-12:
[openmsp430.tar.gz](https://github.com/The-OpenROAD-Project/alpha-release/files/3838127/openmsp430.tar.gz)

Hi
I kept entire flow constant, except that I am using a new design, just to test the flow. Attaching design above. 

Actually, I downloaded this design from opencores.org, modified omsp_clock_gate.v to point to CLKGATETST_X1 and ran make commands

It works well till CTS, but "make route" gives below error:
ERROR: pin _17760_/CK is connected to more than one net
make: *** [results/nangate45/openMSP430/route.guide] Error 255

Can you please take a look?

The goal here was to review how does it infer and place clock gates. Since yosys was not able to infer clock gates, I did the above modification (while I also requested reddit yosys to help on how to infer clock gates)

Thanks

tajayi, 2019-11-13:
Hi @kunalg123,
Unfortunately, the current version of our CTS tool has a few limitations. It only supports a single clock net that must be driven directly by a top level port (i.e. no clock gates, or buffers in between)

kunalg123, 2019-11-13:
I see
Should I keep this issue open to track clock gating?

tajayi, 2019-11-13:
Actually let's keep this open pending when we publish the a "Known Isssues" or Limitations document that explicitly mentions this

tspyrou, 2019-11-25:
We are also working on a new version of CTS. I would like to see it tried on this example when it is ready.

closed 2026-09-09 by maliberty