The-OpenROAD-Project/alpha-release #49: FLUTE error - at least 2 pins in position (0, 0), layer 3

kunalg123, 2019-11-18:
Hi,
I did synth, cts in a different flow, then dumped out a def/verilog using standalone RePlAce and passed it to open-road routing flow, by saving it as 4_cts.def and 4_cts.v. The def already has clock tree built (attaching the testcase). 

While routing, flute gives the below error and comes out. 
Is it happening because I am running the flow from between in unconventional ?? Is the error expected? Do you need some more files/data from me? Please let me know

ERROR: at least 2 pins in position (0, 0), layer 3
ERROR: at least 2 pins in position (0, 0), layer 3
ERROR: at least 2 pins in position (0, 0), layer 3
ERROR: at least 2 pins in position (0, 0), layer 3
ERROR: at least 2 pins in position (0, 0), layer 3
ERROR: at least 2 pins in position (0, 0), layer 3
ERROR: at least 2 pins in position (0, 0), layer 3
ERROR: at least 2 pins in position (0, 0), layer 3

[routing_issue.tar.gz](https://github.com/The-OpenROAD-Project/alpha-release/files/3856713/routing_issue.tar.gz)

tajayi, 2019-11-19:
I opened the def file in a def viewer. My guess here is that all of the io pins are unplaced (or placed on the same location (0,0). So there are several overlapping pins.

tajayi, 2019-12-02:
Closing this issue. Please feel free to re-open if the issue still persists

closed 2019-12-02 by tajayi